sparkfun/SparkFun_u-blox_GNSS_v3 #27: RTCM over UART

forstner-m, 2023-05-31:
Hello,

I am trying to send over UART RTCM data. Unfortunately, only defective packages arrive at the ZED-F9P. I checked this with the u-center. Here is the command: `gnss_serial.pushRawData(data, counter, false);`

When I send the same data over I2C, the transfer works fine. `gnss_i2c.pushRawData(data, counter, false);`

The UART is set correctly because other commands work fine. I suspect that the `pushRawData` function for UART is not working properly. Please check this...

Thanks very much!

Best regards
Michael

PaulZC, 2023-05-31:
Hi Michael (@forstner-m),

Sincere thanks for reporting this!

It looks like a simple dumb mistake...

The Serial version of ```pushRawData``` attempts to push all of the data in one go:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/91c7e9d7bcbe298b82d2c0889923d0ea23bbcffd/src/u-blox_GNSS.cpp#L6093

Except ```writeBytes``` accepts a maximum of 255 bytes (```uint8_t```) and NTRIP data can be _much_ larger than that:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/91c7e9d7bcbe298b82d2c0889923d0ea23bbcffd/src/sfe_bus.cpp#L446

I'll investigate, fix and test tomorrow...

Very best wishes,
Paul

PaulZC, 2023-06-01:
Hi Michael,

I now have a fix for this. The fix is to split serial ```pushRawData``` into small chunks _and_ call ```checkUbloxSerial``` between chunks to avoid data loss. If you are pushing a large amount of data, previously any data received during the push could have been lost - if it overflowed the microcontroller's internal RX buffer. Calling ```checkUbloxSerial``` between chunks prevents this.

I will be releasing version 3.0.15 of the library in a few minutes. Please test it when you have time.

In your code, please call ```gnss_serial.pushRawData(data, counter);``` or ```gnss_serial.pushRawData(data, counter, true);```

Best wishes,
Paul

closed 2023-06-01 by PaulZC

forstner-m, 2023-06-01:
Hello Paul,

thank you for the implementation. The function is working. In the next days I will write you new features.

Best regards,
Michael

commit b0f0eb52 (2023-05-31): Fix #27