sparkfun/SparkFun_u-blox_GNSS_v3 #30: Antenna settings

forstner-m, 2023-06-04:
Hello,

at the moment there are no functions and to change the antenna settings. Commands like this must therefore be used:

`gnss_i2c.setVal8(0x10a3002f, 1);                               //Enable short antenna detection flag
gnss_i2c.setVal8(0x10a30031, 1);                               //Enable open antenna detection flag
gnss_i2c.setVal8(0x10a3002e, 1);                               //Enable active antenna voltage control flag`

It would be good to implement these functions.

Thank you!

Best regards,
Michael

PaulZC, 2023-06-04:
Hi Michael @forstner-m ,

If I add these functions, they will simply be 'helper' functions for the setVal methods. It is not clear to me how much value they will add?

Instead of using the hexadecimal numbers, you can use the key names. Please see:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/pull/19

You can also combine the setVal's into one multi-setVal. Please see:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/main/examples/VALGET_and_VALSET/Example6_multiSetVal_and_GetVal_Templates/Example6_multiSetVal_and_GetVal_Templates.ino

Best wishes,
Paul

forstner-m, 2023-06-04:
This is just a suggestion. I would like the feature, but you do not have to implement it.
Unfortunately, not all hexadecimal numbers are defined and therefore I have to use them.

PaulZC, 2023-06-04:
Hi Michael,

Please see:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/eacf8284c9b7c33143e4893dc8caf8a5d5bbbd32/src/u-blox_config_keys.h#L149-L164

If any are missing, let me know and I will add them.

Best wishes,
Paul

PaulZC, 2023-06-04:
Here is an example of how to use the MON-HW callback:

https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/a6ed3a6fd0baa8c352b7cb1ba9471df9b3f25dbf/Firmware/RTK_Surveyor/Begin.ino#L766

https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/a6ed3a6fd0baa8c352b7cb1ba9471df9b3f25dbf/Firmware/RTK_Surveyor/Begin.ino#L759-L767

https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/a6ed3a6fd0baa8c352b7cb1ba9471df9b3f25dbf/Firmware/RTK_Surveyor/Rover.ino#L251-L254

forstner-m, 2023-06-04:
Hello Paul,

thank you for the help! A custom function would be better, but setVal8 is also usable.

Best regards
Michael

PaulZC, 2023-11-14:
Hi Michael (@forstner-m ),

After thinking about this more, I am going to close this request - marking it as "won't do". The methods would simply be helper methods, making the u-blox configuration keys easier to access. The library is bloated enough already, without adding extra bloat! ;-)

Apologies - and best wishes,
Paul

forstner-m, 2023-11-14:
Hello Paul,

it is okay that you closed the ticket. I have found another way to solve this.

Best regards,
Michael