sparkfun/SparkFun_u-blox_GNSS_v3 #31: UBX-NAV-SAT access

forstner-m, 2023-06-04:
Hello,

I can get the data from UBX-NAV-SAT via `getNAVSAT()`, but access to the data is not possible. Either I cannot find the function or it does not exist. I ask for explanation or implementation.

Thanks you!

Best regards,
Michael

PaulZC, 2023-06-04:
Hi Michael @forstner-m ,

Please see:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/main/examples/Callbacks/CallbackExample5_NAV_SAT/CallbackExample5_NAV_SAT.ino

Best wishes,
Paul

closed 2023-06-04 by PaulZC

forstner-m, 2023-06-04:
Is a access without Callback also possible?

PaulZC, 2023-06-05:
Hi Michael,

Please try:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/release_candidate/examples/Basics/Example33_NAVSAT/Example33_NAVSAT.ino

Best wishes,
Paul

forstner-m, 2023-06-05:
Hello Paul,

thanks for the example That is easier than I thought. But there is a problem:

When I set this: `gnss_i2c.setAutoNAVSAT(true) ` the fuction `gnss_i2c.getNAVSAT()` always returns true. Is this correct?

Thank you!

Best regards,
Michael

PaulZC, 2023-06-06:
Hi Michael,

The solution is to set ```myGNSS.packetUBXNAVSAT->moduleQueried = false;``` after you have read the NAV SAT data.

```moduleQueried``` will be set to ```true``` again when fresh data arrives.

Alternatively, you can call ```myGNSS.flushNAVSAT();``` . It does the same thing:

```
  myGNSS.setNavigationFrequency(1); //Produce one solution per second

  myGNSS.setAutoNAVSAT(true); //Enable periodic NAV SAT messages
}

void loop()
{
  if (myGNSS.getNAVSAT()) // Check for new NAV SAT data
  {
    myGNSS.flushNAVSAT(); // Clear the moduleQueried flag to mark the data as 'stale'
```

Best wishes,
Paul

forstner-m, 2023-06-06:
Hello Paul,

this code works fine. Thanks!

Best regards,
Michael

yashmulgaonkar, 2024-01-29:
NAV_SAT data is severely delayed / slow, with a refresh every almost 15seconds between valid data.
Get the following error over I2C with an ESP32S3 running I2C at both 100khz and 400khz.

```
13:30:08.160 -> Sending: CLS:NAV ID:0x35 Len: 0x0 Payload:
13:30:08.160 -> sendCommand: Waiting for No ACK response
13:30:08.160 -> checkUbloxI2C: 70 bytes available
13:30:08.160 -> checkUbloxI2C: 5 bytes available
13:30:08.299 -> checkUbloxI2C: 32767 bytes available
13:30:09.125 -> Checksum failed: checksumA: 0 checksumB: 6 rollingChecksumA: 93 rollingChecksumB: 92
13:30:10.107 -> Checksum failed: checksumA: 0 checksumB: 0 rollingChecksumA: 26 rollingChecksumB: 34
13:30:11.342 -> processUBX: buffer overrun detected! activePacketBuffer: 0 maximum_payload_size: 3068
13:30:11.342 -> waitForNoACKResponse: CLS/ID match but failed CRC after 3198 msec
13:30:11.389 -> 
13:30:11.389 -> Sending: CLS:NAV ID:0x35 Len: 0x0 Payload:
13:30:11.389 -> sendCommand: Waiting for No ACK response
13:30:11.669 -> checkUbloxI2C: 32767 bytes available
13:30:12.122 -> Checksum failed: checksumA: 89 checksumB: 0 rollingChecksumA: 229 rollingChecksumB: 60
13:30:13.121 -> Checksum failed: checksumA: 25 checksumB: 27 rollingChecksumA: 27 rollingChecksumB: 54
13:30:14.118 -> Checksum failed: checksumA: 29 checksumB: 83 rollingChecksumA: 185 rollingChecksumB: 194
13:30:14.759 -> waitForNoACKResponse: CLS/ID match but failed CRC after 3376 msec
```