sparkfun/SparkFun_u-blox_GNSS_v3 #32: Receive GSA and GSV

forstner-m, 2023-06-09:
Hello,

with the function `getLatestNMEAGNGGA` it is possible to get the GGA sentence. I cannot find a function for GSA and GSV. Is it possible to implement that two sentences.

Thank you!

Best regards,
Michael

PaulZC, 2023-06-17:
Hello Michael,

Apologies for the slow reply - I have been away on vacation.

The issue here is that the module can output _multiple_ GSA and GSV messages in a single navigation cycle. The library would need to be updated to include storage for multiple messages, and provide methods / functions to: tell you how many messages are stored; and to read them one at a time. The term "latest" does not really apply here - there can be multiple "latest" messages.

The best way forward is for you to write your own GSV / GSA parser by overwriting the weak function ```processNMEA``` . Please see [Example2](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/main/examples/Basics/Example2_NMEAParsing/Example2_NMEAParsing.ino) for more details.

Best wishes,
Paul

closed 2023-06-17 by PaulZC

forstner-m, 2023-06-17:
Hello Paul,

in the meantime I have implemented a similar function. So the issue is closed.

Best regards,
Michael