sparkfun/SparkFun_u-blox_GNSS_v3 #39: RTK Corrections unavailable when using RTK Express

divyansh2681, 2023-11-13:
### Subject of the issue
Describe your issue here. If you reference a datasheet please specify which one and in which section (ie, the protocol manual, section 5.1.2). Additionally, screenshots are easy to paste into github.

### Your workbench
* What development board or microcontroller are you using? ESP32
* What version of hardware or breakout board are you using? ESP32
* How is the breakout board wired to your microcontroller? I2C
* How is everything being powered? MicroUSB
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.
```
#include <SparkFun_u-blox_GNSS_v3.h> 
UBX_NAV_STATUS_data_t checkRTKstatus;
Serial.println(checkRTKstatus.fixStat.bits.diffCorr);
```

### Expected behavior
Tell us what should happen
Since I have connected the RTK Express to the GPS and powered it on, I should be receiving the RTK Connections. But when I check the value of `diffCorr` variable (https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/05eea9878f1e91575781d86b82642847fccce2cf/src/u-blox_structs.h#L189C9-L189C9), I always get 0.

### Actual behavior
Tell us what happens instead
The value of `diffCorr` should be 1 when RTK Corrections are available.

PaulZC, 2023-11-14:
Hi Divyansh (@divyansh2681 ),

I have had a quick look at the [source code](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/05eea9878f1e91575781d86b82642847fccce2cf/src/u-blox_GNSS.cpp#L3337) and everything seems OK there. So I think there must be another reason why ```diffCorr``` is not changing state.

Please tell us more about how you have connected the RTK Express to the GPS:
* Which GPS are you using? Is it RTK-capable?
* How are the two connected? Which port on the GPS are you using to receive the corrections?
* Have you checked your wiring? TX->RX, GND->GND?
* Have you confirmed that the Express is outputting corrections? Are the internal switches set correctly?
* Are they both using the same baud rate?
* Does the GPS port have RTCM-In enabled?

Please attach photos here showing the wiring.

Best wishes,
Paul

divyansh2681, 2023-11-15:
> Hi Divyansh (@divyansh2681 ),
>
> I have had a quick look at the [source [...]

Thank you for the response, Paul.

Here are the answers to your questions:

- Which GPS are you using? Is it RTK-capable https://www.amazon.com/gp/product/B0B9TTSN2V/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

- How are the two connected? Which port on the GPS are you using to receive the corrections?
The RTK Express in Base Station mode is connected to the GPS using 2 telemetry radios

- Have you checked your wiring? TX->RX, GND->GND?
I did check the SCL/SDA pins.

- Have you confirmed that the Express is outputting corrections? Are the internal switches set correctly?
I confirmed that it says transmitting, and that the green light is solid while the red light is blinking

- Are they both using the same baud rate?
Both radios are using the same frequency 915MHz

- Does the GPS port have RTCM-In enabled?
I am not sure what this thing means. I am looking into the RTCM examples and figure this out. 

[image]
[image]
[image]
[image]

PaulZC, 2023-11-15:
Hi Divyansh (@divyansh2681 ),

I can't see clearly from your first photo, the cable goes out of shot, but it looks like you have the Radio connected to the GPS-RTK USB port using a USB cable? That is not going to work. You need to provide the Radio with power and connect its TX pin to UART1 RX or UART2 RX on the GPS-RTK.

Please see this tutorial: https://learn.sparkfun.com/tutorials/setting-up-a-rover-base-rtk-system#rover-setup

Scroll down to: https://cdn.sparkfun.com/assets/learn_tutorials/1/3/6/2/Radio_attached_to_ZED-F9P.jpg

I am going to close this because this is an issue with your hardware connections, not an issue with this library. If you need more help, please ask a question on our GPS Forum: https://forum.sparkfun.com/viewforum.php?f=116

Best wishes,
Paul

divyansh2681, 2023-11-18:
Thank you for your help Paul.