sparkfun/SparkFun_u-blox_GNSS_v3 #42: Data is not stored in BBR after myGNSS.powerOff() (software backup mode is not working)

charlienicholsonr, 2023-11-30:
Hello, all!
I am experiencing a weird issue after putting my u-blox MAX M10S GPS module into software backup mode using the function powerOff(). It is not storing time or location data from the previous period of being on. It is almost like it is reseting every time I put it into backup mode. This screenshot from the integration manual says that it stores the relevant data:
[image]
And this screenshot says that it retains BBR:
[image]
But, this screenshot seems contradicting, saying that the RAM is erased:
[image]
Reading futher in the inerface manual, I noticed this. Is the Sparkfun library sensing a reset message every time it "wakes up" from software backup mode?
[image]
I am using a homemade SAMD21G18A microcontroller to interface with the GPS module. The GPS module has power supplied to the V_IO and VCC pins, with nothing supplied to the V_BCKP pin, as specified. The u-blox integration manual can be found using this [link](https://content.u-blox.com/sites/default/files/MAX-M10S_IntegrationManual_UBX-20053088.pdf) (section 3.6.3/3.6.3.2).

PaulZC, 2023-11-30:
Hi Charlie (@charlienicholsonr ),

We'll help you if we can. But, to me, this sounds like an issue with the way you are powering the GNSS and/or using the PMREQ functions.

Please re-read section 4.1 in the integration manual. Are you sure leaving V_BCKP disconnected is the right thing to do, in your case?

RESET_N is the GNSS hardware reset pin. How do you have RESET_N and EXTINT connected? Are you keeping RESET_N high? Are you attempting to use EXTINT to wake the GNSS?

The library has three reset methods (```hardReset```, ```softwareResetGNSSOnly```, ```softwareEnableGNSS```) which all use the common method ```cfgRst```. Internally, ```cfgRst``` isn't called anywhere else, so I don't think the library is responsible for your reset.

In summary, I don't think the library is to blame. I think this is more likely to be a misunderstanding about the sleep state and the wake source.

Best,
Paul

charlienicholsonr, 2023-11-30:
I appreciate your quick response! It very much could be my ignorance of the software backup mode, and not the library's fault. 

According to section 3.6.3.2 of the integration manual, V_BCKP should be left open unless other wise needed:
[image]
I have tried connecting V_BCKP to VCC but to no avail, in case you were wondering.

I have EXTINT connected to GND and I am not trying to wake up the module with it. I have left RESET_N unconnected, but I will run a quick test in a moment with it pulled high.

charlienicholsonr, 2023-11-30:
Okay, did a test and pulling RESET_N high doesn't solve my problem. Btw, this is the serial output:
```
17:05:15.365 -> Sending: CLS:CFG ID:0x8B Len: 0x8 Payload: 0 0 0 0 1 0 71 10
17:05:15.365 -> sendCommand: Waiting for ACK response
17:05:15.431 -> checkUbloxI2C: 27 bytes available
17:05:15.431 -> Incoming: Size: 9 Received: CLS:CFG ID:0x8B Len: 0x9 Payload: 1 0 0 0 1 0 71 10 1
17:05:15.431 -> packetCfg now valid
17:05:15.431 -> packetCfg classAndIDmatch
17:05:15.431 -> Incoming: Size: 2 Received: CLS:ACK ID:0x1 Len: 0x2 Payload: 6 8B
17:05:15.431 -> packetCfg now valid
17:05:15.431 -> packetAck now valid
17:05:15.431 -> packetCfg classAndIDmatch
17:05:15.431 -> packetAck classAndIDmatch
17:05:15.431 -> waitForACKResponse: valid data and valid ACK received after 105 msec
17:05:15.431 -> getVal: sendCommand returned: Data Received
17:05:15.431 -> Date/Time: 
17:05:15.431 -> Sending: CLS:NAV ID:PVT Len: 0x0 Payload:
17:05:15.431 -> sendCommand: Waiting for No ACK response
17:05:15.530 -> checkUbloxI2C: 469 bytes available
17:05:15.563 -> Incoming: Size: 92 Received: CLS:NAV ID:PVT Len: 0x5C Payload: E0 2E 0 0 E5 7 3 7 0 0 C F0 FF FF FF FF 0 0 0 0 0 0 24 0 0 0 0 0 0 0 0 0 0 0 0 0 98 BD FF FF FF FF FF FF 0 76 84 DF 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 20 4E 0 0 80 A8 12 1 F 27 0 0 5C 40 56 2F 0 0 0 0 0 0 0 0
17:05:15.563 -> packetCfg now valid
17:05:15.563 -> packetCfg classAndIDmatch
17:05:15.595 -> waitForNoACKResponse: valid data with CLS/ID match after 164 msec
17:05:15.595 -> 2021-3-7 0:0:12
17:05:15.595 -> -- Powering off module for 20s --
17:05:15.595 -> Powering off for 20000 ms
17:05:15.595 -> 
17:05:15.595 -> Sending: CLS:0x2 ID:0x41 Len: 0x8 Payload: 20 4E 0 0 2 0 0 0
17:05:15.595 -> sendCommand: Waiting for No ACK response
17:05:15.726 -> checkUbloxI2C: 34 bytes available
17:05:15.924 -> checkUbloxI2C: 35 bytes available
17:05:16.714 -> waitForNoACKResponse: TIMEOUT after 1100 msec. No packet received.
17:05:26.697 -> -- waking up module via pin 11 on your microcontroller --
17:05:28.707 -> Date/Time: 
17:05:28.707 -> Sending: CLS:NAV ID:PVT Len: 0x0 Payload:
17:05:28.707 -> sendCommand: Waiting for No ACK response
17:05:28.805 -> checkUbloxI2C: 469 bytes available
17:05:28.838 -> Incoming: Size: 92 Received: CLS:NAV ID:PVT Len: 0x5C Payload: E8 3 0 0 E5 7 3 7 0 0 1 F0 FF FF FF FF 0 0 0 0 0 0 24 0 0 0 0 0 0 0 0 0 0 0 0 0 98 BD FF FF FF FF FF FF 0 76 84 DF 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 20 4E 0 0 80 A8 12 1 F 27 0 0 5C 40 56 2F 0 0 0 0 0 0 0 0
17:05:28.838 -> packetCfg now valid
```

closed 2023-12-01 by PaulZC

PaulZC, 2023-12-01:
Hi Charlie,

You can also check the time validity with ```getTimeValid```, ```getDateValid```, ```getConfirmedTime```, ```getConfirmedDate``` and ```getTimeFullyResolved```. You could wait for those to go true before powering off, instead of waiting for a fixed period.

All the best,
Paul

charlienicholsonr, 2023-12-02:
Hi Paul,

I don't know whether to open this issue again, but I am encountering some issues after integrating the GNSS code with the rest of my code (which, by the way, is a LoRa tracker). I am having the same problems as before, but this time I am checking if the time data is valid with `getConfirmedTime` before continuing on to the rest of the loop. 

The annoying thing is, I got my code working, and then I stupidly decided to change something, and it stopped working. It probably is just a silly error on my part, especially seeing how much I modified your (working) code. 

Here's my sketch:
```
#include <SPI.h> // Needed for LoRa.
#include <LoRa.h> // LoRa.
#include <Wire.h> // Needed for I2C to GPS module.
#include <SparkFun_u-blox_GNSS_v3.h> // http://librarymanager/All#SparkFun_u-blox_GNSS_v3.

// Settings:
#define comment "XXXXXX"
#define baudRate 115200 // SerialUSB baud rate.
#define syncWord 0xFB // Only other devices with this sync word can receive your broadcast.
#define ssPin 38 // Arduino pin that connects to LoRa module. AKA csPin. 
#define resetPin 5 // Arduino pin that connects to LoRa module.
#define DIO0Pin 2 // Arduino pin that connects to LoRa module.
#define frequency 433E6 // Frequency of your LoRa module.
#define spreadingFactor 12 // Choose this based on this https://forum.arduino.cc/t/what-are-the-best-settings-for-a-lora-radio/449528.
#define bandwidth 62.5E3  // Choose this based on this https://forum.arduino.cc/t/what-are-the-best-settings-for-a-lora-radio/449528.
#define firstAwakeForSecs 120 // You need to have enough time for the GPS to sync before the first sleep, or else the RAM data is unvalidated and is not used by the GPS
#define sleepForSecs 5000 // How many milliseconds do you want to sleep for?
#define devMode // Comment this out when you're ready to do the actual flight to save even more power.
// #define testCoord // Uncomment this to use test coordinates instead of actual coordinates.

struct __attribute__((packed)) dataStruct {
  float lat;
  float lon;
  long alt;
  long sats;
  long speed;
  long course;
  long txCount = 0;
  char text[50] = comment;
} transmittingData;

#ifdef testCoord
  float lat = 42.316651;
  float lon = -71.366030;
  int alt = 96;
  int sats = 9;
  int speed = 60;
  int course = 360;
#endif

SFE_UBLOX_GNSS gps;

void setup() {
  #ifdef devMode
    SerialUSB.begin(baudRate); 
    while (!SerialUSB); // Wait until SerialUSB is all good.
    SerialUSB.println("LoRa Tracker v2.1");
  #endif

  LoRa.setPins(ssPin, resetPin, DIO0Pin); // SS, reset, and DIO0. Has to be before LoRa.begin().

  if (!LoRa.begin(frequency)) {
    #ifdef devMode
      SerialUSB.println("Starting LoRa failed!");
    #endif
    while(1);
  }

  LoRa.setSyncWord(syncWord); 
  LoRa.setSpreadingFactor(spreadingFactor);
  LoRa.setSignalBandwidth(bandwidth);
  LoRa.crc();
[... 107 lines cut ...]
  gps.powerOffWithInterrupt(sleepForSecs, 0, true); // No (additional) wakeup sources. force = true

  #ifdef devMode
    SerialUSB.print("Transmitting with a packet size of: ");
    SerialUSB.println(sizeof(transmittingData));
  #endif
  transmit(transmittingData); // This is blocking and takes a tad bit more than 5 seconds.
  #ifdef devMode
    SerialUSB.println("Done transmitting. Sleeping LoRa.");
  #endif
  LoRa.sleep(); // The LoRa module wakes up automatically when the SPI interface is active.
}

void transmit(const struct __attribute__((packed)) dataStruct &data) {
  // Send the packet over LoRa.
  LoRa.beginPacket();
  LoRa.write((byte *)&data, sizeof(data));
  LoRa.endPacket();
}
```

My code needs the GPS module to be on for the least amount of time possible to save power. Note that there isn't an MCU sleep or delay in the code because I'm relying on the transmit function to block the sketch for more than 5 seconds. One thing that could be causing issues in my code is the fact that I have a LoRa module and a GPS module on the same breadboard, and RF interference may be occurring.

Thanks for all your help, 
Charlie

PaulZC, 2023-12-02:
Hi Charlie,

TL;DR: If you move ```getConfirmedTime``` inside the ```getPVT``` loop, it will (probably) start working again.

The ```confirmedTime``` flag is one bit of ```flags2``` of the PVT (Position Velocity Time) message.
```getConfirmedTime``` will call ```getPVT``` internally if it needs to.
If ```getPVT``` times out, because the module is still doing its cold start and doesn't want to communicate right now, then ```getConfirmedTime``` will return whatever random bit is in the memory that holds the PVT message (or will hold the PVT message when one arrives!).

The methods like ```getConfirmedTime``` are not truly safe. Strictly, I should have written them like this:

```bool getConfirmedTime(bool *confirmedTime, uint16_t maxWait)```

with the confirmedTime bit being written into the bool pointed to by ```*confirmedTime```. The method itself would (only) return true if it knows the confirmedTime is valid - i.e. the ```getPVT``` was successful, or had previously been successful.

```getPVT``` only returns true if a PVT message was successfully received. If you call ```getConfirmedTime``` inside the ```if (getPVT() == true)``` loop, you know the returned value is valid (not stale, not random).

All the best,
Paul

PS. Later, I did start to add safe versions of (previously) unsafe methods like this. Strictly, I should do the same thing _everywhere_... Including ```getConfirmedTime```. Sorry about that...

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/05eea9878f1e91575781d86b82642847fccce2cf/src/u-blox_GNSS.h#L435-L436

PaulZC, 2023-12-02:
Hi Charlie,

You don't need the ```gps.factoryDefault()``` and ```gps.hardReset()```. Those were just to get the module into a known state for the sleep test. You can safely comment or delete those.

After a cold start, ```getPVT``` will keep returning false until the module has completed its cold start and is ready to communicate. This takes a couple of seconds. You need to keep calling ```getPVT``` until it returns true, then you can safely call ```getConfirmedTime```. So try something like:

```
  static bool firstAwake = true;

  if (firstAwake) {
    bool confirmedTime = false;
    while (!confirmedTime) {
      if (myGNSS.getPVT()) {
        mySerial.print("Date/Time: ");
        mySerial.print(myGNSS.getYear());
        mySerial.print("-");
        mySerial.print(myGNSS.getMonth());
        mySerial.print("-");
        mySerial.print(myGNSS.getDay());
        mySerial.print(" ");
        mySerial.print(myGNSS.getHour());
        mySerial.print(":");
        mySerial.print(myGNSS.getMinute());
        mySerial.print(":");
        mySerial.println(myGNSS.getSecond());
        confirmedTime = myGNSS.getConfirmedTime();
      }
      else {
        mySerial.println("No PVT data received. Retrying...");
      }
    }
  }

  firstAwake = false;
```

Best wishes,
Paul

charlienicholsonr, 2023-12-02:
Hi Paul, 

The code now waits for the time to be validated before moving on. Although it now waits, after sleep the time comes out unvalidated. Is there something else I should check before going to sleep?

Cheers, 
Charlie

PaulZC, 2023-12-03:
Hi Charlie,

I think that's all as expected. The GNSS knows that its internal clock will drift a little while it is asleep and not receiving satellite signals. So it's expected that it will need to re-sync / re-validate when it wakes.

Cheers,
Paul

ushalev001, 2024-11-13:
Hi,
I have a similar application where powe consumption is critical. I was using the power_off function, but I noticed that the power of the GPS (MAX-M10M) even though reduced dramaticaly, still consumes about 2-3mA. it is supposed to consume 47uA according to the spec. I was strugling with this for months - any idea why? the code that I am using for the test is the exact Sparkfun example for power_off

ushalev001, 2024-11-14:
> Hi @ushalev001<https://github.com/ushalev001> ,
>
> There are a few pieces to this:
>
> Use a passive antenna to minimise the current draw.
> If you need to use an active antenna, you can use UBLOX_CFG_HW_ANT_CFG_PWRDOWN_POL to disable the VCC_RF while the module is asleep. see: https://community.sparkfun.com/t/ublox-max-m10-ublox-max-10s-external-antenna-circuit/47229
>
> If you have a power LED, disconnect it.
>
> If you have a PPS LED, disconnect it. With the SparkFun MAX-M10S board, I see current leaking through the PPS LED while the module is asleep.
>
> Be aware that the EXTINT pin may leak current. Holding the EXTINT pin low adds about 240uA to the sleep current.
>
> You need to use the 16-byte version of the RXM-PMREQ message, and you need to make sure the "force" bit is set. Using this library, the function is powerOffWithInterrupt. powerOff uses the 8-byte version which does not support the force bit. Code example is below.
>
> I hope this helps,
> Paul

Hi Paul,
Thank you so much for the quick response. There is a lot to digest … let me try and revert and you gave me hope as I am struggling with this issue for more than a year now.
Regards,
us

SFE_UBLOX_GNSS_SERIAL myGNSS;

// define a digital pin capable of driving HIGH and LOW

#define WAKEUP_PIN 4

// Possible GNSS interrupt pins for powerOffWithInterrupt are:

// VAL_RXM_PMREQ_WAKEUPSOURCE_UARTRX  = uartrx

// VAL_RXM_PMREQ_WAKEUPSOURCE_EXTINT0 = extint0 (default)

// VAL_RXM_PMREQ_WAKEUPSOURCE_EXTINT1 = extint1

// VAL_RXM_PMREQ_WAKEUPSOURCE_SPICS   = spics

// These values can be or'd (|) together to enable interrupts on multiple pins

void wakeUp() {

  Serial.print("-- waking up module via pin " + String(WAKEUP_PIN));

  Serial.println(" on your microcontroller --");

  digitalWrite(WAKEUP_PIN, LOW);

  delay(1000);

  digitalWrite(WAKEUP_PIN, HIGH);

  delay(1000);

  digitalWrite(WAKEUP_PIN, LOW);

}

void setup() {

  pinMode(WAKEUP_PIN, OUTPUT);

  digitalWrite(WAKEUP_PIN, LOW);

  Serial.begin(115200);

  while (!Serial); //Wait for user to open terminal

  Serial.println("SparkFun u-blox Example");

  Serial.println("Hit any key to begin");

  while (Serial.available())

    Serial.read();

  while (!Serial.available())

    ;

  Serial1.begin(38400); // Change to 9600 if needed

  myGNSS.enableDebugging(); // Enable debug messages

  if (myGNSS.begin(Serial1) == false) //Connect to the u-blox module using Serial1

  {

    Serial.println(F("u-blox GNSS not detected. Please check wiring. Freezing."));

    while (1);

  }

  // Powering off for 20s, you should see the power consumption drop.

  Serial.println("-- Powering off module for 20s --");

  //myGNSS.powerOff(20000); // PMREQ 8-byte version

  myGNSS.powerOffWithInterrupt(20000); // PMREQ 16-byte version. Defaults to VAL_RXM_PMREQ_WAKEUPSOURCE_EXTINT0, true

  //myGNSS.powerOffWithInterrupt(20000, VAL_RXM_PMREQ_WAKEUPSOURCE_EXTINT0, true); // PMREQ 16-byte version. Set wakeup source. Set force bit.

  //myGNSS.powerOffWithInterrupt(20000, 0, true); // PMREQ 16-byte version. No wakeup source. Set force bit.

  //myGNSS.powerOffWithInterrupt(20000, 0, false); // PMREQ 16-byte version. No wakeup source. Clear force bit.

  delay(10000);

  // After 10 seconds wake the device via the specified pin on your microcontroller and module.

  wakeUp();

}

void loop() {

  //Do nothing

}

ushalev001, 2024-11-14:
Hi Paul,

I have implemented the following:
--------------------------------

1 – using Passive antenna
2 – Calling myGNSS.setVal8(UBLOX_CFG_HW_ANT_CFG_PWRDOWN_POL, 1);
3 – calling myGNSS.powerOffWithInterrupt(20000);
4 – The LED shuts-off immediately when calling the powerOff so I did not disassemble it
5 - I am not using EXTINT pin
6 – The module starts working when I supply power to it, so there is no additional WAKEUP pin

Unfortunately, I am still getting the same ~3mA power consumption
Interesting enough, when I was using poweroff() the current went down to 10mA, and only when I called it twice (in a raw)  it reduced to ~3mA

This is the Code I am using for the test:
------------------------------------------------

#include <SparkFun_u-blox_GNSS_v3.h> //http://librarymanager/All#SparkFun_u-blox_GNSS_v3
SFE_UBLOX_GNSS_SERIAL myGNSS;

#define mySerial Serial1
void setup()
{
  Serial.begin(115200);
  while (!Serial); //Wait for user to open terminal

  int8_t rx=D4,tx=D5;
  do {
    mySerial.begin(115200,SERIAL_8N1,rx,tx); // u-blox F9 and M10 modules default to 38400 baud. Change this if required
    if (myGNSS.begin(mySerial) == true){
      Serial.println("115200 success");
      break;
      }
    delay(100);
  } while(1);

  Serial.println("GNSS serial connected");

  //myGNSS.setUART1Output(COM_TYPE_UBX); //Set the UART port to output UBX only
  //myGNSS.saveConfiguration(); //Save the current settings to flash and BBR

    myGNSS.enableDebugging(); // Enable debug messages
    Serial.println("\n----     myGNSS.setVal8(UBLOX_CFG_HW_ANT_CFG_PWRDOWN_POL, 1); ");
    myGNSS.setVal8(UBLOX_CFG_HW_ANT_CFG_PWRDOWN_POL, 1);
    delay(2000);
    Serial.println("---- calling myGNSS.powerOffWithInterrupt(20000); ");
    myGNSS.powerOffWithInterrupt(20000);
    delay(21000);
    Serial.println("Test Ended");
    while(1);
}

This is the terminal output:
----------------------------------

08:49:51.737 -> 115200 success
08:49:51.737 -> GNSS serial connected
08:49:51.737 ->
08:49:51.737 -> ----     myGNSS.setVal8(UBLOX_CFG_HW_ANT_CFG_PWRDOWN_POL, 1);
08:49:51.737 ->
08:49:51.737 -> Sending: CLS:CFG ID:0x8A Len: 0x9 Payload: 0 3 0 0 34 0 A3 10 1
08:49:51.737 -> sendCommand: Waiting for ACK response
08:49:51.835 -> Incoming: Size: 2 Received: CLS:ACK ID:0x1 Len: 0x2 Payload: 6 8A
08:49:51.835 -> packetCfg now valid
08:49:51.835 -> packetAck now valid
08:49:51.835 -> packetCfg classAndIDmatch
08:49:51.835 -> packetAck classAndIDmatch
08:49:51.835 -> waitForACKResponse: no data and valid ACK after 122 msec
08:49:53.847 -> ---- calling myGNSS.powerOffWithInterrupt(20000);
08:49:53.847 -> Powering off for 20000 ms
08:49:53.847 ->
08:49:53.847 -> Sending: CLS:0x2 ID:0x41 Len: 0x10 Payload: 0 0 0 0 20 4E 0 0 6 0 0 0 20 0 0 0
08:49:53.847 -> sendCommand: Waiting for No ACK response
08:49:53.847 -> Incoming: Size: 92 Received: CLS:NAV ID:PVT Len: 0x5C Payload: IGNORED
08:49:53.847 -> packetCfg now valid
08:49:53.847 -> Incoming: Size: 92 Received: CLS:NAV ID:PVT Len: 0x5C Payload: IGNORED
08:49:53.847 -> packetCfg now valid
08:49:53.847 -> Incoming: Size: 92 Received: CLS:NAV ID:PVT Len: 0x5C Payload: IGNORED
08:49:53.847 -> packetCfg now valid
08:49:53.879 -> Incoming: Size: 92 Received: CLS:NAV ID:PVT Len: 0x5C Payload: IGNORED
08:49:53.879 -> packetCfg now valid
08:49:54.932 -> waitForNoACKResponse: TIMEOUT after 1100 msec. No packet received.
08:50:15.942 -> Test Ended

Regards,
uz

ushalev001, 2024-11-14:
Hi Paul,

One more thing I should have emphasized more.
This implementation is using MAX-10M, not MAX-10S

PaulZC, 2024-11-14:
Hi Uz,

I only have MAX-M10S...

If ```powerOff``` reduces the current to 10mA, and ```powerOffWithInterrupt``` reduces it to 3mA, it suggests everything is working. With ```powerOff``` on the -M10S SparkFun board, the current reduced to ~3mA. With ```powerOffWithInterrupt```, it reduced to ~350uA. When I disconnected EXTINT, it reduced to ~110uA. And some / most of that will be the current charging the small back-up battery.

I think you need to be very methodical, and try to find where the current is going. It may be leaking through GPIO pins with pull-up / pull-down.

Have you checked that antenna voltage VCC RF is off while the module is asleep? Measure the voltage to be sure...

If you share your schematic, I will try to find time to take a look.

Best,
Paul

ushalev001, 2024-11-14:
Hi Paul,Thanks, it helps!I thought VCC-RF is identical to VCC, no logic between them in the chip. Is there a command to control VCC-RF?Regards,uz

ushalev001, 2024-12-11:
Hi Paul,
the issue is indeed related to the LNA continue to draw current, so fixing this solved the issue - Many thanks.
Now to complete the project, I need to have the AOP to work. I send the module to poweroff every 15 minutes and what I see is that when it wakes up it takes about 4-10 seconds to get to the accuracy that I need (ubxDataStruct->hAcc). Then before sending the module to poweroff I am doing 2 things:

1 - save the data (do I really need to do this?)
2 - making sure AOP is in rest

like this:
//==========================
      if (!((cfgStat = myGNSS.saveConfiguration()))) {. // Is this necessary?
        Serial.print("myGNSS.saveConfiguration(); Failed ");
        Serial.println(cfgStat);
      }

      while (myGNSS.getAOPSTATUSstatus()) {. // Making sure AOP is not busy when poweroff
        Serial.print("wait on getAOPSTATUSstatus ");
        delay(100);
      }

      Serial.println("getAOPSTATUSstatus cleared");

      myGNSS.powerOffWithInterrupt((SLEEP_INTERVAL - GPS_WAKE_BEFORE_ESP) * 1000, VAL_RXM_PMREQ_WAKEUPSOURCE_UARTRX);

//=========================

Now the issue I have is that it takes getAOPSTATUSstatus() about 4 seconds before it returns false.... which is almost the same time that takes to the module to get his position right.

what is the reason it takes so long? and is there a way to make it faster? for example should I tell the module to stop navigate once I have reached my target accuracy?

Many thanks in advance
uz

ushalev001, 2024-12-15:
what is the power consumption in power off? is the sparkfun module supports the feature to turn off the LNA when in poweroff?

ushalev001, 2024-12-15:
Paul,
Thanks, that’s help a lot. I tried with some other off-the-shelf module and it does work!

So, I will switch to MAX-M10S, and use a passive antenna (MAX-M10S already has an internal LNA), will it then turn-off the LNA when I call powerOffWithInterrupt without adding any MOSFET to gate the power?

Regards,
uz

PaulZC, 2024-12-16:
Hi uz,

Apologies for the slow reply - I was away on vacation.

If you use a passive antenna, you should be able to achieve the 47uA sleep current. With an active antenna, you will need to add the MOSFET circuit to switch the antenna power.

It is a long time since I wrote the AOPSTATUS examples. But, from what I remember, the module needs a few seconds to collect the AssistNow Autonomous data and for the AOP ```status``` flag to go idle. Again from what I remember, for the fastest start up you need to provide the GNSS with time information using ```myGNSS.setUTCTimeAssistance```. Please see Example3_AssistNowAutonomous_DatabaseWrite for more details.

I hope this helps,
Paul

ushalev001, 2024-12-16:
Thanks Paul, not issue, I am running statistics to optimize my solution, and it anyhow takes time.

So,
I can’t see why an external LNA to the m10S be required… the antenna is a ceramic integrated antenna so one LNA in my 30 years of experience in RF (4G/5G, not GNSS….) should be sufficient and an additional LNA may push a higher signal but won’t improve the signal to noise ration. Do you think I am wrong on this?

So, some background of what I am trying to achieve,

The application is small battery powered (100Ah) device, it is a tracker device aimed to locate small animals (size of a cat) . So weight and size also matters. They are slow moving (like human walking speed), and never get further away than a1000 meters radius. The accuracy of the location (2D only) should be not less than 8 meters, so it doesn’t have to be great accuracy, but 15 meters is too much…

The SW logic is to wake-up the device every 15 minutes, get a 2D location and go back to a sleep mode. So, in average anything more than 8 second per iteration will start to squeeze the validity of the solution because the battery will get drain too quickly.

In general, logically everything is working well but I am failing to properly optimize the solution to achieve the goals.

The way I am testing is, I know the location of the device
So, here are the parameters I have been playing with:
// ------------------------------------------- Code ----------------------------------------------------------
myGNSS.setDynamicModel(DYN_MODEL_PEDESTRIAN, VAL_LAYER_RAM_BBR);
configureGNSSFor2DMode(); // This is a function I wrote to set to 2D mode, not sure I did it correctly, and I am not sure it helps at all….
myGNSS.saveConfigSelective(VAL_CFG_SUBSEC_IOPORT); // do I need to do this every time? I am not certain what it actually saves…

// I am using call backs, but do not print anything from the callback functions themselves.
myGNSS.setAutoPVTcallbackPtr(&printPVTdata);  // Enable automatic NAV PVT messages with callback to printPVTdata
myGNSS.setAutoNAVSATcallbackPtr(&newNAVSAT);  // Enable automatic NAV SAT messages with callback to newNAVSAT

// not sure this helps but saw it in many examples
myGNSS.setUART1Output(COM_TYPE_UBX);  //Set the UART1 port to output UBX only (turn off NMEA noise)

// using only GPS and SBAS seemed like the fastest and most efficient way, not sure why adding the others hurts the modem to get to accuracy faster
// SBAS seems to help

myGNSS.enableGNSS(true, SFE_UBLOX_GNSS_ID_GPS);       // Only enable GPS for proper software backup mode
  myGNSS.enableGNSS(true, SFE_UBLOX_GNSS_ID_SBAS);      // Enable SBAS
  myGNSS.enableGNSS(false, SFE_UBLOX_GNSS_ID_GALILEO);  // Disable Galileo
  myGNSS.enableGNSS(false, SFE_UBLOX_GNSS_ID_BEIDOU);   // Disable BeiDou
  myGNSS.enableGNSS(false, SFE_UBLOX_GNSS_ID_IMES);     // Disable IMES
  myGNSS.enableGNSS(false, SFE_UBLOX_GNSS_ID_QZSS);     // Disable QZSS
  myGNSS.enableGNSS(false, SFE_UBLOX_GNSS_ID_GLONASS);  // Disable GLONASS

// I’ve noticed that faster reporting frequency helps, again I am not sure why….
  myGNSS.setNavigationFrequency(GPS_REPORT_FREQ);

// The loop part is printing constantly, I spare you the printing part of the code, it is straight forward….
// the # of SVs
// how many are locked
// what is the actual horizontal error (I know the true location)
// what is the estimated Horizontal error

//After the following conditions are met: , I spare you the printing part of the code , it is straight forward ….
// Horizontal error estimation < 5 meters
// at least 8 SVs are seen
// AOP is not busy

// the loops ends with sending the module to poweroff, and ESP32 deep sleep. The poweroff will end 5 seconds before the ESP is waking up.
myGNSS.powerOffWithInterrupt((SLEEP_INTERVAL - GPS_WAKE_BEFORE_ESP) * 1000, VAL_RXM_PMREQ_WAKEUPSOURCE_UARTRX);
esp_sleep_enable_timer_wakeup((SLEEP_INTERVAL - DELAY_FOR_DEBUG) * uS_TO_S_FACTOR);
esp_deep_sleep_start();

// ----------------------------------------- end of code ------------------------

The strange thing I see is that sometimes the actual accuracy is almost perfect, 7 locked SVs, but the Horizonal estimation get stuck for maybe even 10 full seconds in 11meters error and then on one step goes to 1 meter…. So a waste of 10 seconds because there was something wrong with the estimation.
Not sure how to fix this.

I am not sure setting it to 2D helps at all. Should it? Very little documentation about it
I am not sure that the update frequency and the callbacks method for getting the info is helping for better results, there is not documentation about it
My Baud rate is 115200, is this an issue?
Lastly, I am waking up the module 5 seconds before the ESP32 wakes up, I thought that would be good from power consumption view point because it alwas takes at least 5 seconds to get to this accuracy. Is that a good practice? Or the Module gets confused and its configuration when waking up needs to be done every time it wakes up.

I am not getting great results, so my question is if those methods above are the right ones or is there a better way to get an lower average time for a specific accuracy.

So,
So many questions, it’s a long journey for me. The u-blox documentation is sub-optimal for this use case, everyone is concentrating these days on drones…

Best Regards, and thanks in advance
Uz

PaulZC, 2024-12-17:
Hi Uz,

Thanks for the update. I'm afraid I'm not going to be of much help here. I have experimented a little with "powerOff", but have never devoted as much time to it as you have. 

I did find a u-blox white paper which may help:

https://www.u-blox.com/sites/default/files/Low-power-tracking-applications_WhitePaper_UBX-21014570%281%29.pdf

The "pico-ballooning" guys must have solved this. That's what they do: sleep, wake, get a fix as fast as possible, send it via radio chirp, and go back to sleep again. Usually using solar power.

Maybe, with some careful searching, you can find the code you need?

If there are any features we can add to this library to help you, please let us know.

Best,
Paul

ushalev001, 2024-12-17:
Thanks Paul, you have been a great help already. I am familiar with this white paper. It’s too high level but thanks for sharing.Do you know if there any way to get u- blox guys to answer some of those questions?Regarding the library, I think two features could be useful 1- setting the 2D fix mode2- setting criteria to use only satellites that are above certain degrees above the horizon to be used (there are some other interesting criterias)Thanks Paul,Regards,uz

https://www.u-blox.com/sites/default/files/Low-power-tracking-applications_WhitePaper_UBX-21014570%281%29.pdf