sparkfun/SparkFun_u-blox_GNSS_v3 #43: NTRIP Connection dropped

rekabuk, 2023-12-19:
### Subject of the issue
Using the SparkFun "NTRIPClient_With_CGA_Callback_Serial" example code I can successfully open a connection to the rtk2go base station, but after about ~4 times round the main loop(), the connection is dropped - how can I determine why it was dropped and how do I stop it being dropped?

### Your workbench
* What development board or microcontroller are you using?   "DOIT ESP32 DEVKIT V1"
* What version of hardware or breakout board are you using?  "No breakout - Serial connection (Serial1) to u-Blox ZED-F9P GNSS receiver board"
* How is the breakout board wired to your microcontroller?      
* How is everything being powered?                                         "PC USB, My NTRIP server based on your example (same hardware) works a treat and has been running for many days."
* Are there any additional details that may help us help you?   "I am using SparkFun u-blox GNSS v3 (3.0.16)"

### Steps to reproduce
I just substituted my parameters into your example code, the only change I made was some debug to determine if "keypressed()" or "processConnection()" was causing the connection to be 'closed' - it is processConnection()

### Expected behavior
Code should not drop the connection to the rtk2go NTRIP server

### Actual behavior
The connection is "dropped"

rekabuk, 2023-12-19:
Here is the redacted serial print out

Connecting to local WiFi.
WiFi connected with IP: 192.168.1.65
Connecting to the NTRIP caster...
Opening socket to rtk2go.com
Connected to rtk2go.com : 2101
Requesting NTRIP Data from mount point XXXXXXX
Sending credentials: XXXXXXX:YYYYYYY
serverRequest size: 130 of 513 bytes available
Sending server request: 
GET /XXXXXXXXXXX HTTP/1.0
User-Agent: NTRIP SparkFun u-blox Client v1.0
Authorization: Basic ccccccccccccccccccccccccccc=

Connected to: rtk2go.com
Connected to the NTRIP caster! Press any key to disconnect...
Pushed 2048 RTCM bytes to ZED
Pushing GGA to server: $GPGGA,102601.00,5106.66067,N,00255.27234,W,2,05,2.75,20.5,M,48.9,M,,0136*7B
Pushed 2048 RTCM bytes to ZED
Pushed 2048 RTCM bytes to ZED
Pushed 1961 RTCM bytes to ZED
Connection dropped!
Close - Connectrion fail
Closing the connection to the NTRIP caster...
Disconnected!
Press any key to reconnect...

rekabuk, 2023-12-19:
Things a far more stable (but not completely stable) if I delete the  line:

 myGNSS.setNMEAGPGGAcallbackPtr(&pushGPGGA); 
 
 If I add this line the connection always drops after two pushes of data to the receiver.

PaulZC, 2023-12-19:
Hi Andrew (@rekabuk ),

Can you please enable "Show Timestamp" in the Serial Monitor and run it again. I'd like to see the timing of the disconnect.

Which version of the Espressif arduino-esp32 board package are you using? Which board do you selected?

I wonder if the Serial baud rate is causing issues. Are you able to try I2C instead? Or can you increase the GNSS UART baud rate to (at least) 115200 baud? Try ```myGNSS.setSerialRate(115200);``` and then restart / re-begin at 115200.

Cheers,
Paul

rekabuk, 2023-12-19:
Using I2C would be a pain, but I can certainly try a faster baud rate to the receiver.

rekabuk, 2023-12-19:
That was a bit more of a struggle than it should have been, but I ended up with the GNSS UART running at 230400 and the Example seems to be running without dropping the connection.

Thank-you @PaulZC  :-)

closed 2023-12-19 by PaulZC