sparkfun/SparkFun_u-blox_GNSS_v3 #45: Control the rate of data from NTRIP Caster

rekabuk, 2023-12-20:
#Subject of the issue
Using the SparkFun "NTRIPClient_With_CGA_Callback_Serial" example code I can successfully open a connection to the rtk2go base station, but my processor gets flooded with data, no matter how big I make the rtcmData buffer, it always overflows and hits the "break" statement. How can I control the rate at which data arrives from the caster (Assuming I have it correct and that's where the data is coming from?).
I have increased the serial BAUD rate to the Z9P to 203400,  but things still fall over eventually.

    What development board or microcontroller are you using? "DOIT ESP32 DEVKIT V1"
    What version of hardware or breakout board are you using? "No breakout - Serial connection (Serial2) to u-Blox ZED-F9P GNSS receiver board"
    How is the breakout board wired to your microcontroller?
    How is everything being powered? "PC USB, My NTRIP server based on your example (same hardware) works a treat and has been running for many days."
    Are there any additional details that may help us help you? "I am using SparkFun u-blox GNSS v3 (3.1.00)"

#Steps to reproduce
I just substituted my parameters into your example code, the only change I made was some debug to determine if "keypressed()" or "processConnection()" was causing the connection to be 'closed' - it is processConnection()

#Expected behavior
Code should not drop the connection to the rtk2go NTRIP server

#Actual behavior
The connection is "dropped"

PaulZC, 2023-12-20:
Hi Andrew (@rekabuk ),

Your NTRIP Server sets how much data the Caster will send to the Client. How many messages do you have enabled, at at what rate? Try reducing the rate and/or the number of messages...

Best wishes,
Paul

rekabuk, 2023-12-20:
I had a feeling you might says that, I haven't found how to do it on rtk2go yet :-(

Andrew

PaulZC, 2023-12-20:
You need to adjust your NTRIP Server code - to set which messages are sent and how often. RTK2go blindly forwards whatever it receives, you can't configure it...

rekabuk, 2023-12-20:
NTRIP server (base)   --> ~200 Bps -----> rtk2go.com caster ------> ~24KBps -----> NTRIP Client (Rover)

PaulZC, 2023-12-20:
Are you sure you've set the ```mountPoint``` and ```casterPort``` correctly - in secrets.h? Also ```casterUser``` needs to be a valid email address - although me@there.com also works... ;-)

Which NTRIP Rev did you request when you created the mount point? I'd suggest Rev1. If you selected Rev2, that might explain it?

rekabuk, 2023-12-20:
Hmmm, although the client is "receiving" and pushing data to the Z9P, rtk2go says no data is being sent to any clients - sound like a good place to start looking.
Andrew

rekabuk, 2023-12-20:
If I set an invalid password, username etc, the client still thinks it has connected correctly and I get the same "connection dropped" behaviour after a short time :-(

rekabuk, 2023-12-21:
It seems I was using the wrong username, so the NTRIP client wasn't actually connecting, despite all the responses coming back saying everything was OK!

And my caster is Rev1 :-)

Thanks again for your help @PaulZC 

Andrew

PaulZC, 2023-12-21:
Cheers Andrew,
Paul

closed 2023-12-21 by PaulZC