sparkfun/SparkFun_u-blox_GNSS_v3 #49: SV data updates very slow, over 10 seconds between valid data.

yashmulgaonkar, 2024-01-29:
### Subject of the issue
NAV_SAT data is severely delayed / slow, with a refresh every almost 15seconds between valid data.
Get the following error over I2C with an ESP32S3 running I2C at both 100khz and 400khz.

* Running Example 33 for NAV_SAT.
* Running on ESP32S3 dev kit.
* GPS: UBlox NEO-M9N over I2C

### Here's a snippet of the debug output:
```
13:30:08.160 -> Sending: CLS:NAV ID:0x35 Len: 0x0 Payload:
13:30:08.160 -> sendCommand: Waiting for No ACK response
13:30:08.160 -> checkUbloxI2C: 70 bytes available
13:30:08.160 -> checkUbloxI2C: 5 bytes available
13:30:08.299 -> checkUbloxI2C: 32767 bytes available
13:30:09.125 -> Checksum failed: checksumA: 0 checksumB: 6 rollingChecksumA: 93 rollingChecksumB: 92
13:30:10.107 -> Checksum failed: checksumA: 0 checksumB: 0 rollingChecksumA: 26 rollingChecksumB: 34
13:30:11.342 -> processUBX: buffer overrun detected! activePacketBuffer: 0 maximum_payload_size: 3068
13:30:11.342 -> waitForNoACKResponse: CLS/ID match but failed CRC after 3198 msec
13:30:11.389 -> 
13:30:11.389 -> Sending: CLS:NAV ID:0x35 Len: 0x0 Payload:
13:30:11.389 -> sendCommand: Waiting for No ACK response
13:30:11.669 -> checkUbloxI2C: 32767 bytes available
13:30:12.122 -> Checksum failed: checksumA: 89 checksumB: 0 rollingChecksumA: 229 rollingChecksumB: 60
13:30:13.121 -> Checksum failed: checksumA: 25 checksumB: 27 rollingChecksumA: 27 rollingChecksumB: 54
13:30:14.118 -> Checksum failed: checksumA: 29 checksumB: 83 rollingChecksumA: 185 rollingChecksumB: 194
13:30:14.759 -> waitForNoACKResponse: CLS/ID match but failed CRC after 3376 msec
```

PaulZC, 2024-01-29:
Hi Yash (@yashmulgaonkar ),

I have Example33 running on a Feather ESP32S3 board with the SparkFun NEO-M9N SMA breakout connected on Qwiic I2C. Everything is working well (see below).

I suspect you are seeing I2C bus errors from a lack of pull-ups? Or maybe the combined pull-up resistance is too low? The bytes available showing as 32767 is very suspicious.

Please check if your dev kit has pull-ups on the SDA and SCL pins. If you are using a SparkFun NEO-M9N breakout, please check if the I2C jumpers are still closed / connected.

Best wishes,
Paul

```
22:38:03.152 -> New NAV SAT data received. It contains data for 31 SVs.
22:38:03.152 -> GPS     2   
22:38:03.152 -> GPS     3   =========================
22:38:03.152 -> GPS     8   
22:38:03.152 -> GPS     10  
22:38:03.152 -> GPS     14  ======================================
22:38:03.152 -> GPS     17  ============================================
22:38:03.152 -> GPS     19  ======================
22:38:03.152 -> GPS     21  ==========================
22:38:03.152 -> GPS     22  ===============================
22:38:03.152 -> GPS     24  
22:38:03.152 -> GPS     32  =============================
22:38:03.152 -> SBAS    123 
22:38:03.152 -> SBAS    127 
22:38:03.152 -> SBAS    136 
22:38:03.152 -> Galileo 21  ===================================
22:38:03.152 -> BeiDou  9   ============
22:38:03.152 -> BeiDou  11  ===========================
22:38:03.152 -> BeiDou  19  ===========================
22:38:03.152 -> BeiDou  22  ===============================
22:38:03.152 -> BeiDou  23  =============================
22:38:03.152 -> BeiDou  34  ===============================
22:38:03.152 -> GLONASS 1   =====================
22:38:03.152 -> GLONASS 2   ============================
22:38:03.152 -> GLONASS 3   ========================
22:38:03.152 -> GLONASS 4   
22:38:03.152 -> GLONASS 9   =================================
22:38:03.152 -> GLONASS 10  ===========================
22:38:03.152 -> GLONASS 11  
22:38:03.152 -> GLONASS 17  ==============
22:38:03.152 -> GLONASS 18  =================
22:38:03.152 -> GLONASS 19  ======================
22:38:04.089 -> 
22:38:04.089 -> New NAV SAT data received. It contains data for 31 SVs.
22:38:04.089 -> GPS     2   
22:38:04.089 -> GPS     3   =========================
22:38:04.089 -> GPS     8   
22:38:04.089 -> GPS     10  
22:38:04.089 -> GPS     14  ======================================
22:38:04.089 -> GPS     17  ============================================
22:38:04.089 -> GPS     19  ======================
22:38:04.089 -> GPS     21  ==========================
22:38:04.089 -> GPS     22  ===============================
22:38:04.089 -> GPS     24  
22:38:04.089 -> GPS     32  =============================
22:38:04.089 -> SBAS    123 
22:38:04.089 -> SBAS    127 
22:38:04.089 -> SBAS    136 
22:38:04.089 -> Galileo 21  ===================================
22:38:04.089 -> BeiDou  9   
22:38:04.089 -> BeiDou  11  ===========================
22:38:04.089 -> BeiDou  19  ===========================
22:38:04.089 -> BeiDou  22  ===============================
22:38:04.089 -> BeiDou  23  =============================
22:38:04.089 -> BeiDou  34  ===============================
22:38:04.089 -> GLONASS 1   =====================
22:38:04.089 -> GLONASS 2   ============================
22:38:04.089 -> GLONASS 3   =======================
22:38:04.089 -> GLONASS 4   
22:38:04.089 -> GLONASS 9   =================================
22:38:04.089 -> GLONASS 10  ===========================
22:38:04.089 -> GLONASS 11  
22:38:04.089 -> GLONASS 17  ==============
22:38:04.089 -> GLONASS 18  =================
22:38:04.089 -> GLONASS 19  ======================
```

PaulZC, 2024-01-29:
Is it possible the module is in a weird state, with a high navigation rate? Please try adding the following to reset the module, at line 44 - after the ```myGNSS.begin()``` :

```
myGNSS.factoryDefault();
delay(5000);
```

yashmulgaonkar, 2024-01-29:
I do have I2C pullups (4.7K) on the ESP32 dev kit as well as the native pullups (2.2K) on the Sparkfun board.
That gives me an equivalent of a little under 1.5K pullups on each of the SDA/ SCL lines.

I have tried the factory defaults line before, and that did not help either.

PaulZC, 2024-01-29:
OK. Please try opening the pull-up jumpers on the GNSS board.

yashmulgaonkar, 2024-01-29:
Nope no dice - same behavior - tried with one other ESP32S3 dev kit and two other NEO-M9N modules I have on hand. So definitely not a module issue.

PaulZC, 2024-01-30:
Hi Yash, (@yashmulgaonkar ),

OK - thanks. This has to be something to do with your setup, but I will help if I can.

1. Please post a link to the exact dev board you are using
2. Please post a photo of your setup showing the wires connecting the two boards
3. Do you have any other boards connected?
4. Have you made _any_ changes to Example33? If so, please post them here.
5. Which version of the Arduino IDE are you using?
6. Which version of the ESP32 board package?
7. Which board have you selected?
8. Have you made any changes to the board settings (USB, partition scheme, PSRAM, etc.)? If so, please post them here.

Best,
Paul

PaulZC, 2024-02-09:
Closing... Please re-open if you need more help with this.

Best wishes,
Paul

closed 2024-02-09 by PaulZC