sparkfun/SparkFun_u-blox_GNSS_v3 #54: Relative Position NED not returning proper data over I2C.

Supernova1114, 2024-04-08:
### Subject of the issue
I am able to view Relative Position NED data from the U-Blox U-Center application via USB. I have the Relative Position NED automatically sending packets over the UBX protocol, and can view those packets being sent. When connected over I2C, I am able to receive positional data from the simple GetPosition Arduino example, however, when I test the RelativePositionInformation example, the packet returns zero for all data points.

### Your workbench
* What development board or microcontroller are you using? Arduino Mega 2560
* What version of hardware or breakout board are you using? SparkFun RTK-2 Zed-F9P breakout.
* How is the breakout board wired to your microcontroller? Wired over I2C using Qwiic connectors, then to Dupont wires into the Arduino.
* How is everything being powered? GPS modules are powered over the I2C Qwiic connector to Arduino. Arduino powered via USB.
* Note: Something that is missing from the image given is that I have a single wire for serial TX to RX (moving base GPS, to moving rover GPS) connected to to the header pins I added to the breakout boards. This is used for the movement of RTCM3 correction data.
* Another note: GPS antennas were also connected during testing, which are not included in the image.

[image]

### Steps to reproduce
1. Connect rover GPS that produces RelPosNED solutions via USB to computer with U-Blox U-center application.
2. Double check that Relative Position NED is working within u-center. (Gives valid solution) and receives heading information.
3. Disconnect USB cable.
4. Run Arduino code and see that zeros are output from every data point. The only data that is set correctly is gnssFixOK. After a good while, diffSoln will also set itself to true.

Stripped down code from the Zed-F9P > Example5_RelativePositioningInformation.ino example:

```
#include <Wire.h> //Needed for I2C to GNSS

#include <SparkFun_u-blox_GNSS_v3.h> //http://librarymanager/All#SparkFun_u-blox_GNSS_v3
SFE_UBLOX_GNSS myGNSS;

// Callback
void printRELPOSNEDdata(UBX_NAV_RELPOSNED_data_t *ubxDataStruct)
{
  Serial.println();
  Serial.println("New RELPOSNED data received:");

  Serial.print("relPosLength (m): ");
  Serial.println(((double)ubxDataStruct->relPosLength / 100) + ((double)ubxDataStruct->relPosHPLength / 10000), 4); // Convert cm to m
  Serial.print("relPosHeading (Deg): ");
  Serial.println((double)ubxDataStruct->relPosHeading / 100000); // Convert deg * 1e-5 to degrees

  Serial.print("gnssFixOk: ");
  if (ubxDataStruct->flags.bits.gnssFixOK == true)
    Serial.println("x");
  else
    Serial.println("");

  Serial.print("diffSolution: ");
  if (ubxDataStruct->flags.bits.diffSoln == true)
    Serial.println("x");
  else
    Serial.println("");

  Serial.print("relPosValid: ");
  if (ubxDataStruct->flags.bits.relPosValid == true)
    Serial.println("x");
  else
    Serial.println("");

  Serial.print("carrier Solution Type: ");
  if (ubxDataStruct->flags.bits.carrSoln == 0)
    Serial.println("None");
  else if (ubxDataStruct->flags.bits.carrSoln == 1)
    Serial.println("Float");
  else if (ubxDataStruct->flags.bits.carrSoln == 2)
    Serial.println("Fixed");

  Serial.print("isMoving: ");
  if (ubxDataStruct->flags.bits.isMoving == true)
    Serial.println("x");
  else
    Serial.println("");
}

void setup()
{
  delay(1000);
  
  Serial.begin(115200);
  Serial.println("u-blox Base station example");

  Wire.begin();
  Wire.setClock(400000); //Increase I2C clock speed to 400kHz

  while (myGNSS.begin(Wire, 0x43) == false) //Connect to the u-blox module using Wire port
  { 
    Serial.println(F("u-blox GNSS not detected at default I2C address. Please check wiring. Retrying..."));
  }

  myGNSS.setAutoRELPOSNEDcallbackPtr(&printRELPOSNEDdata); // Enable automatic NAV RELPOSNED messages with callback to printRELPOSNEDdata
}

void loop()
{
  myGNSS.checkUblox(); // Check for new RELPOSNED data
  myGNSS.checkCallbacks();
}
```

### Expected behavior
Return proper data from RelPosNED.

### Actual behavior
All data points return zero from RelPosNED.

PaulZC, 2024-04-08:
Hi @Supernova1114 ,

The Mega is a 5V board. The ZED is 3.3V. You will get errors in the I2C signals due to the voltage difference. You need a level shifter: https://www.sparkfun.com/products/15439

Also, please check you are connecting your red wire to the Mega 3.3V breakout pin. (I can't see the pin clearly in your photo.) You may damage the ZED if you connect Qwiic 3.3V to 5V.

I hope this helps,
Paul

Supernova1114, 2024-04-08:
That is a whoops by my part, I think I was indeed using 5V. Thankfully the boards don't seem to be damaged yet. I will get back to y'all near the end of the day to see if this solves the issue.

PaulZC, 2024-04-08:
Also, you have two ZEDs on the same I2C bus. Have you set the address of one to 0x43? Have you saved the address in ZED flash memory? (Are you sure the addresses are correct - and not causing bus collisions?)

Best,
Paul

PaulZC, 2024-04-09:
Closing this as it seems clear it is a hardware issue. Please re-open if you need more help.
Best wishes,
Paul

closed 2024-04-09 by PaulZC

Supernova1114, 2024-04-11:
This ended up fixing the issue thank you!

PaulZC, 2024-04-11:
Thanks for the update. Glad that's working for you...