sparkfun/SparkFun_u-blox_GNSS_v3 #55: getModuleInfo() fails for UBX-M10050-KB

amnn314, 2024-04-18:
### Subject of the issue
I found that getModuleInfo() returns false for my UBX-M10050-KB device. I have also found out that it happened because the payload did't contain 'MOD=' substring.

Here's the debug output I get, showing that there's no such substring, while FWVER and PROTVER are present:

```
Sending: CLS:MON ID:0x4 Len: 0x0 Payload:
sendCommand: Waiting for No ACK response
Incoming: Size: 160 Received: CLS:MON ID:0x4 Len: 0xA0 Payload: 46 57 56 45 52 3D 53 50 47 20 35 2E 31 30 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 50 52 4F 54 56 45 52 3D 33 34 2E 31 30 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 47 50 53 3B 47 4C 4F 3B 47 41 4C 3B 42 44 53 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 53 42 41 53 3B 51 5A 53 53 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
packetCfg now valid
packetCfg classAndIDmatch
waitForNoACKResponse: valid data with CLS/ID match after 233 msec
```

I decoded it using the following Python one-liner:
```
>>> s = "46 57 56 45 52 3D 53 50 47 20 35 2E 31 30 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 50 52 4F 54 56 45 52 3D 33 34 2E 31 30 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 47 50 53 3B 47 4C 4F 3B 47 41 4C 3B 42 44 53 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 53 42 41 53 3B 51 5A 53 53 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0"
>>> "".join([chr(int(x, 16)) for x in s.split(' ')])
'FWVER=SPG 5.10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00PROTVER=34.10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00GPS;GLO;GAL;BDS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00SBAS;QZSS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>>
```

### Your workbench
Actual device used: https://www.hglrc.com/products/hglrc-m100_mini-gps

### Expected result
Probably, DevUBLOXGNSS::getModuleInfo() could be fixed by not expecting all three items to be present.

PaulZC, 2024-04-18:
Hi @amnn314 ,

Thank you for reporting this.

I was worried initially that you might have a cloned part. But, looking at the M10 SPG 5.10 Interface Description, it says that the _MOD_ is only reported _if available_. I guess the UBX-M10050-KB _chip_ does not report _MOD_.

[image]

I will correct this in the next few minutes.

Best wishes,
Paul

PaulZC, 2024-04-18:
Hi @amnn314 ,

Please try v3.1.5. Let me know if it resolves your issue.

Best wishes,
Paul