sparkfun/SparkFun_u-blox_GNSS_v3 #57: Error in connecting in serial mode

a-3isa, 2024-05-31:
i am getting this 
[image]

when i am try to connect to gps02-ubx using esp32c3    

`#include "HardwareSerial.h"
#include <SparkFun_u-blox_GNSS_v3.h>  //http://librarymanager/All#SparkFun_u-blox_GNSS_v3
SFE_UBLOX_GNSS myGNSS;

HardwareSerial mySerial(1);

long lastTime = 0;  //Simple local timer. Limits amount if I2C traffic to u-blox module.

struct gps_readings {
  long lat, lon, alt;
};

void gps_init() {
  Serial.println("GPS initializing....");

  do {
    Serial.println("GNSS: trying 38400 baud");

    mySerial.begin(38400, SERIAL_8N1, RX1_PIN, TX1_PIN);

    myGNSS.enableDebugging();  // Uncomment this line to enable helpful debug messages on Serial
    myGNSS.connectedToUART2();  // This tells the library we are connecting to UART2 so it uses the correct configuration keys

    if (myGNSS.begin(mySerial) == true) break;

    delay(100);
    Serial.println("GNSS: trying 9600 baud");
    mySerial.begin(9600, SERIAL_8N1, RX1_PIN, TX1_PIN);
    if (myGNSS.begin(mySerial) == true) {
      Serial.println("GNSS: connected at 9600 baud, switching to 38400");
      myGNSS.setSerialRate(38400);
      delay(100);
    } else {
      //myGNSS.factoryDefault();
      delay(2000);  //Wait a bit before trying again to limit the Serial output
    }
  } while (1);
  // Serial.println("GNSS serial connected");
  Serial.println("The GPS is successfully initialized.");
  Serial.println("================================================");

  myGNSS.setUART1Output(COM_TYPE_UBX);  //Set the UART port to output UBX only
  myGNSS.saveConfiguration();           //Save the current settings to flash and BBR
}
`

PaulZC, 2024-06-03:
Hi @a-3isa ,

In your code, you are trying to use Serial to communicate with the gps02-ubx module on UART2 (```myGNSS.connectedToUART2();```). But I think the gps02-ubx module does not provide access to UART2, only UART1?

In your screenshot, I think you are using I2C to communicate with the gps02-ubx module? This does not match your code.

To help you further, we need to see a schematic and a photo of how you have connected the gps02-ubx module to the ESP32.

Best wishes,
Paul

PaulZC, 2024-06-05:
Closing... Please reopen if you need more help with this.

closed 2024-06-05 by PaulZC