sparkfun/SparkFun_u-blox_GNSS_v3 #63: Powering off using EXTINT

gauteh, 2024-08-13:
Hi, I am trying to turn of the MAX-M10S using the suggestions here: https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/228

I have enabled both EXTINTBACKUP and EXTINTWAKE, so that the device should be completely in the control of the EXTINT pin and not wake up on any other sources. It does not seem to work yet. Can you spot anything wrong? Maybe I need to configure PSMOO as well?

https://github.com/gauteh/omb-2024-nano/blob/main/src/gnss_manager.cpp#L7:
```
  wdt.restart();
  Serial.println(F("Attempting to configure GPS enable pin.."));

  GnssWire.begin();

  /* turn_gnss_on(); */
  delay(1000); // Give it time to power up
  wdt.restart();

  Serial.println(F("gnss wire started"));
  Serial.flush();

  if (!gnss.begin(GnssWire, 0x42)){
    Serial.println(F("problem starting GNSS"));
    return false;

    // power things down
    delay(500);
  } else{
    Serial.println(F("success starting GNSS"));
  }

  // now we know that we can talk to the gnss
  gnss.setI2COutput(COM_TYPE_UBX); // Limit I2C output to UBX (disable the NMEA noise)
  delay(100);

  // If we are going to change the dynamic platform model, let's do it here.
  // Possible values are:
  // PORTABLE,STATIONARY,PEDESTRIAN,AUTOMOTIVE,SEA,AIRBORNE1g,AIRBORNE2g,AIRBORNE4g,WRIST,BIKE
  if (!gnss.setDynamicModel(DYN_MODEL_STATIONARY)){
    Serial.println(F("GNSS could not set dynamic model"));
  }

  Serial.println(F("configuring..."));
  gnss.newCfgValset(); // Defaults to configuring the setting in RAM and BBR
  gnss.addCfgValset(UBLOX_CFG_PM_EXTINTBACKUP, 1);
  gnss.addCfgValset(UBLOX_CFG_PM_EXTINTWAKE, 1);
  gnss.sendCfgValset(); // Send the packet using sendCfgValset
  Serial.println(F("done."));

  wdt.restart();
  GnssWire.end();

  return true;
}

```

PaulZC, 2024-08-13:
Hi Gaute (@gauteh ),

> It does not seem to work yet.

How are you judging this? Current draw?

Best,
Paul

gauteh, 2024-08-13:
Yes. I have checked that the pin is low. Is there a good way to test?

PaulZC, 2024-08-13:
Thanks Gaute.

Do you see _any_ reduction in the current when the INT pin is low? The current should reduce, but will (probably) still be several mA.

PaulZC, 2024-09-27:
Closing... Please reopen if you need more help with this.

Best,
Paul

closed 2024-09-27 by PaulZC

gauteh, 2024-09-27:
Yes, I realized it is too easy to wake it up again, so I will have to make some power switch.