sparkfun/SparkFun_u-blox_GNSS_v3 #67: AutoPVT with MAX-M10S

gauteh, 2024-09-27:
Hi, should this example work with MAX-M10S? I never get any auto callbacks, I do get PPS etc.

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/main/examples/Example8_PositionVelocityTime_Callback/Example8_PositionVelocityTime_Callback.ino

My code: https://github.com/gauteh/sfy/blob/sfy-rtk/sfy-buoy/sfy-ext-gps/ext-gps-mod/demo_gps.cpp#L68

PaulZC, 2024-09-27:
Hi Gaute (@gauteh ),

Example 8 is working OK for me - on a SparkFun Thing Plus C with a SparkFun MAX-M10S breakout connected via Qwiic:

[image]

Maybe try resetting your module, just in case you have misconfigured the port settings? Add ```myGNSS.factoryDefault(); delay(5000);``` after the ```myGNSS.begin()```. You should only need to do this once.

I'll try your code next.

Best,
Paul

PaulZC, 2024-09-27:
I had to modify your code a little, but it works too. Sounds like this might be a case of PEBCAK? ;-)

Best,
Paul

[image]

```
#include <SparkFun_u-blox_GNSS_v3.h> //http://librarymanager/All#SparkFun_u-blox_GNSS_v3

TwoWire GnssWire(0);
SFE_UBLOX_GNSS gnss;

void getPVT(UBX_NAV_PVT_data_t *ubxDataStruct)
{
    Serial.println();

    Serial.print(F("Time: ")); // Print the time
    uint8_t hms = ubxDataStruct->hour; // Print the hours
    if (hms < 10) Serial.print(F("0")); // Print a leading zero if required
    Serial.print(hms);
    Serial.print(F(":"));
    hms = ubxDataStruct->min; // Print the minutes
    if (hms < 10) Serial.print(F("0")); // Print a leading zero if required
    Serial.print(hms);
    Serial.print(F(":"));
    hms = ubxDataStruct->sec; // Print the seconds
    if (hms < 10) Serial.print(F("0")); // Print a leading zero if required
    Serial.print(hms);
    Serial.print(F("."));
    uint32_t millisecs = ubxDataStruct->iTOW % 1000; // Print the milliseconds
    if (millisecs < 100) Serial.print(F("0")); // Print the trailing zeros correctly
    if (millisecs < 10) Serial.print(F("0"));
    Serial.println(millisecs);
}

void setup() {

  delay(1000);

  Serial.begin(115200);
  
  Serial.println(F("GPS: Initiating GNSS."));

  GnssWire.begin();
  delay(1000); // Give it time to power up

  if (!gnss.begin(GnssWire, 0x42)){
    Serial.println(F("problem starting GNSS"));

    return;

    // power things down
    delay(500);
  } else{
    Serial.println(F("success starting GNSS"));
  }

  uint8_t ok = gnss.setI2COutput(COM_TYPE_UBX); //Turn off NMEA noise
  if (ok) ok = gnss.setI2CInput(COM_TYPE_UBX);
  if (ok) ok = gnss.setUART1Output(0);
  if (ok) ok = gnss.setUART1Input(0);
  if (ok) ok = gnss.setUART2Output(0);
  /* if (ok) ok = gnss.setUART2Input(COM_TYPE_SPARTN); //Be sure SPARTN input is enabled */
  /* if (ok) ok = gnss.setDGNSSConfiguration(SFE_UBLOX_DGNSS_MODE_FIXED); // Set the differential mode - ambiguities are fixed whenever possible */
  if (ok) ok = gnss.setNavigationFrequency(1); //Set output in Hz.
  /* if (ok) ok = myGNSS.setVal8(UBLOX_CFG_SPARTN_USE_SOURCE, 0); // Use "IP" source, not L-Band. We are injecting raw SPARTN, not PMP */
  /* if (ok) ok = myGNSS.setVal8(UBLOX_CFG_MSGOUT_UBX_RXM_COR_I2C, 1); // Enable UBX-RXM-COR messages on I2C */

  // now we know that we can talk to the gnss
  /* delay(100); */

  // If we are going to change the dynamic platform model, let's do it here.
  // Possible values are:
  // PORTABLE,STATIONARY,PEDESTRIAN,AUTOMOTIVE,SEA,AIRBORNE1g,AIRBORNE2g,AIRBORNE4g,WRIST,BIKE
  /* if (!gnss.setDynamicModel(DYN_MODEL_STATIONARY)){ */
  /*   Serial.println(F("GNSS could not set dynamic model")); */
  /* } */

  /* gnss.setAutoPVT(true); */
  gnss.setAutoPVTcallbackPtr(&getPVT); // Enable automatic NAV PVT messages with callback to printPVTdata so we can watch the carrier solution go to fixed
}

void loop() {
  gnss.checkUblox();
  gnss.checkCallbacks();
}
```

closed 2024-09-27 by PaulZC

gauteh, 2024-09-27:
That is usually the case 😊 thanks for checking, I will try with your code.

PaulZC, 2024-09-27:
This might help:

Your code fails at ```if (ok) ok = gnss.setUART2Output(0);``` because the MAX-M10S has no UART2. So the ```if (ok) ok = gnss.setNavigationFrequency(1);``` is not executed because ```ok``` is then false...