sparkfun/SparkFun_u-blox_GNSS_v3 #70: SAM-M10Q PowerOff Issue

pchwalek, 2025-02-10:
Hello, 

Through my testing, I found that the SAM-M10Q needs the full UBX-RXM-PMREQ message sent (i.e., _powerOffWithInterrupt()_) if wanting the device to go to low power mode for a specified number of milliseconds. I have tried _powerOff()_ and the device does acknowledge but it does not go into standby mode. 

Best,
Patrick

PaulZC, 2025-02-10:
Hi Patrick (@pchwalek ),

There is a discussion about ```powerOffWithInterrupt``` [here](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/issues/42#issuecomment-2475932429). ```powerOffWithInterrupt``` sets the ```force``` bit, whereas ```powerOff``` does not.

Is there anything in the library you think we should change? Or are you just sharing your experiences?

Best wishes,
Paul

PaulZC, 2025-04-03:
Closing. Please reopen if you need more help with this.

closed 2025-04-03 by PaulZC