sparkfun/SparkFun_u-blox_GNSS_v3 #73: Feature Request: add lever-arm configuration for F9R

Eric-FR, 2025-03-11:
With the last firmware update of the F9R (FW1.00HPS1.40), Realase note (7 February 2025) tells: 

> 2.4 Lever arm support
In systems where the level of accuracy is comparable to the offsets between the various
reference points like GNSS antenna phase center, IMU reference frame, middle of the vehicle
rear axle, these offsets need to be specified generally referred to as lever arms. In previous
releases of the product, the lever arms were confidential messages reserved for advanced
users. These messages are now publicly available to user and supported by u-center.

I would be very interested in being able to set such parameters, as it is difficult to put the F9R juste above the centre of rear axle.

Integration manual (21-Feb-2025):
>3.2.4.3.3 User-defined lever arm
The lever arm keys CFG-SFCORE-IMU2CRP, CFG-SFODO-IMU2VRP and CFG-SFIMU-IMU2ANT must be interpreted in the following way:
• LA_X: Corresponds to the x-axis component of the lever arm.
• LA_Y: Corresponds to the y-axis component of the lever arm.
• LA_Z: Corresponds to the z-axis component of the lever arm.

Interface description (18-Feb-2025, Protocol version 33.40):
One can find corresponding configurations in
> 6.9.20 CFG-SFCORE: Sensor fusion (SF) core configuration
6.9.21 CFG-SFIMU: Sensor fusion (SF) inertial measurement unit (IMU)
configuration
6.9.22 CFG-SFODO: Sensor fusion (SF) odometer configuration

Eric-FR, 2025-03-12:
Adding configuration Key IDs would be a good starting point. I'm not sure that, at this point, it is worth implementing more integrated function.

Eric-FR, 2025-03-17:
Indeed, one step forward would be to add an option in the RTK Express Plus allowing user to set manually such parameters.

PaulZC, 2025-04-03:
The keys have been added in v3.1.9

Commit: 9d9e2ea70b62723c7c8eecfe5aec74d532326c0e

PaulZC, 2025-04-03:
@Eric-FR :

This is resolved in https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/releases/tag/v3.1.9

Please open a separate issue in the [RTK Firmware Repo](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues) to request support in the RTK Firmware. Thanks!

closed 2025-04-03 by PaulZC