sparkfun/SparkFun_u-blox_GNSS_v3 #74: Feature Request: add rail dynamic for F9R

Eric-FR, 2025-03-12:
With the last firmware update of the F9R (FW1.00HPS1.40), Realase note (7 February 2025) tells:

> 2.1 Rail and tram use case support
Vehicle behaving as rail vehicle and trams can calibrate the inertial navigation system for
applications requiring a meter level of accuracy.

Integration manual:

> 3.1.10.1 Platform settings
Rail vehicles Used for applications with dynamics equivalent to those of trains and trams.

Interface description

> Table 25: Constants for CFG-NAVSPG-DYNMODEL
RAIL 13 Rail vehicles (trains, trams) (not available in all products)

Can the configuration Key value added?

PaulZC, 2025-03-13:
Hi Eric,

Thanks. We'll add this when we add the new lever arm keys (#73 ).

Best,
Paul

PaulZC, 2025-04-03:
This will be included in v3.1.9

Commit: 58d7a1f2f722a086aa74a7b10a8c9cfd1ab1bae0

PaulZC, 2025-04-03:
Resolved in https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/releases/tag/v3.1.9

closed 2025-04-03 by PaulZC